adamtheone/canDrive #2: Possibility to help eachother and share databases

jppw, 2020-11-24:
No issue but an idea.

Possibility to upload and share collected data and mapped ids from diferent vehicles.

adamtheone, 2020-11-28:
Inside the 02_canSniffer_GUI/save folder, those 2 files contain all my decoded packets and ID.
But I agree, it would be good to collaborate easier, so I've created this google sheet, so everybody can add their findings:

https://docs.google.com/spreadsheets/d/1eBKiTwEE6aPnfw2EfSHItLeMz00fJq0Y0L99Ym7fKwU/edit?usp=sharing

I'm adding this link to the README as well!

Arbe-GTC, 2020-12-07:
Hi,

In this post https://github.com/Trueffelwurm/MPi3/issues/1 from @Trueffelwurm you have some data from Astra H. Its very similar to Vectra C.

I have some IDs captured from my Astra from low and mid speed can bus.

This projects are very interestings!

montyx99, 2021-03-23:
Why the google sheet is empty for me? Someone has deleted the content as I see.

adamtheone, 2022-01-23:
I've protected the existing sheets.
If you wish to add a new page write me an email, and I can restrict the editing rights for your email!

xQted, 2024-06-18:
[Audi SQ5 (2016).csv](https://github.com/user-attachments/files/15893010/Audi.SQ5.2016.csv)
[audi_c6_acan_497_01-modded_By_xeQted.dbc.txt](https://github.com/user-attachments/files/15893013/audi_c6_acan_497_01-modded_By_xeQted.dbc.txt)
[ThinkCity.dbc.txt](https://github.com/user-attachments/files/15893024/ThinkCity.dbc.txt)