adamtheone/canDrive #3: Display more than 8 data bytes

morcibacsi, 2020-11-27:
Not really an issue but a suggestion. It would be awesome to have more than 8 places for the data bytes. I know it was mainly written with CAN bus in mind which has a max length of 8 data bytes, however there are some protocols which supports more than that. I spent a lot of time reverse engineering the VAN bus used in Peugeot cars where a message could contain up to 30 bytes of data. If you would extend the interface, this could become an universal sniffing tool.

Nice project, and videos by the way ;)

adamtheone, 2020-11-28:
Thanks for the suggestion, it makes sense. 
It should not be a problem, but would require some ground work, since the max 8 bytes is hardcoded in some places in the code too. I'll keep this idea in mind, and we'll see whether this is a useful feature for others too.

adamtheone, 2020-11-29:
Thanks for the PR, I'll review it in the upcoming days!