adamtheone/canDrive #4: Can't get it to start (after installing)

SketchThis, 2020-11-27:
I'n trying to get CanDrive running in Ubuntu 20. I've done some CAN sniffing with Arduino, but I'm pretty new to Linux and Python... I'm struggling to even get CanDrive to start:

[image]

Is there any kind of a walkthrough or instructions for a new person?

adamtheone, 2020-11-28:
As I'm using windows, I have not seen that error message before, but some forums suggest that should try to reinstall this package:
`sudo apt-get install --reinstall libxcb-xinerama0`

More info:
- https://discuss.pixls.us/t/solved-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found/17677
- https://forum.qt.io/topic/111553/solved-qt-qpa-plugin-could-not-find-the-qt-platform-plugin-xcb-in/15
- https://forum.qt.io/topic/108755/could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found

FrankM54, 2021-02-01:
I had the same error and reinstalling the libxcb resolved the issue.

closed 2021-02-20 by adamtheone