adamtheone/canDrive #5: Integration canDrive with OP-COM Interface

adamslodki, 2020-11-29:
I'am use OP-COM interface to sniffing CAN buses in Opel Vectra C using SocketCAN project and [ opcom.py]( https://bugs.lwpm.eu/task/182) script, but yours software (canDrive) looks great. May be you would like to integrate your canDrive with OP-COM interface.
opcom.py scripts allows software switching beetwen LOW, MEDIUM or HIGH CAN-BUS in Opel cars.

adamtheone, 2022-02-03:
Hello.
I only used the OPCOM dongle with the official OPCOM software to reverse engineer the packets. The dongle and the software were obviously fully compatible with all the 3 busses of my car, so I didn't need to do anything on that front, I was just listening to the packets. With that said, I'm not sure how to control the OPCOM interface. As you could see in the videos, I've always separated the busses by hardware: the arduino could talk to only 1 bus at a time, and my custom hardware included 3 different can drivers in order to communicate with all busses at the same time.
Unfortunately, I don't have free capacity currently to integrate the OPCOM interface into this project, although it would be cool for sure!

zalexzperez, 2024-10-06:
Hi Adam, 

did you ever try getting all the DTCs from the OPCOM data base?
It seems they're encrypted in files for each model/engine combination.
There is a compilation of all the errors on the internet but they don't carry the last two characters in the error code to specifically pinpoint the actual error, like OPCOM does.

adamtheone, 2025-01-06:
Sorry for the late reply! No, I did not try that.

closed 2025-01-06 by adamtheone