adamtheone/canDrive #8: Crashes with

jonchaka, 2021-01-04:
I can only get this to work on the ms-can bus.

To clarify, my vehicle is a Mazda BT50, they use the same running gear as a Ford Ranger. The ms-can is pretty much Ford.
There are 2 buses with ms-can occupying pin 3/11 on the obd port. There is a sync module between the 2 buses, but I don't think it passes over all information. Might only pass over information requested. 

HS-Can runs at 1000
MS-Can runs at 250

While trying the standard (hs) can. Once ignition is on, the pycharm script will exit with "Process finished with exit code -1073741571 (0xC00000FD)". The can system will wake up with key on accessories, it only exits when key is turned to ignition.

Not sure where to start with finding out what is going on. Would be happy to do any debugging.

The arduino output looks normal. Not sure if the higher bit rate is causing the script to exit or not, as the ms-can seems to run fine. Have tried this on 2 different computers to rule out a hardware/memory limitation. I'm the same person that posted on the youtube video.

jonchaka, 2021-01-04:
Ford/Mazda do use extended IDs. Not sure if that is causing it.

adamtheone, 2021-01-21:
Hey jonchaka!

Thanks for reaching out and I'm very sorry for the late reply. 
First of all yes, the gui only supports 11bit ID, but it's just a matter of some lengths checks of some arrays. The arduino and the library supports it completely (IDE). I can try to extend it and make it work, but given my limited free time, it would be really cool if you could help me out with that.

On the other topic: I'm pretty sure it's the high baud-rate. Are you saying that when you open the serial terminal within Arduino IDE, the packets are coming in nicely? If no, you can try to increase the baud rate in the arudino code. If yes, it's the python thread that cannot keep up with the load of incoming packages. What you could do is you can apply some CAN filters within the arduino code OR (just for sanity check) you could try to NOT send every packet from the ardunio. Like send every 2nd packet or something like that. If you've validated the root cause, we can work on that more.

FrankM54, 2021-02-01:
> Ford/Mazda do use extended IDs (29bits). Not sure if that is causing it.

What year did they start?
From what I have seen in North America, it is 11 bit @ 500Kbs.

adamtheone, 2021-02-20:
I've added 29bit ID support in the branch. If you could try it out and leave your feedback, that'd be super useful:
https://github.com/adamtheone/canDrive/tree/29bit_id_support

ccritix, 2021-03-27:
Hi.
Unfortunately, I still get the Process finished with exit code -1073741571 (0xC00000FD) error. I make contact with the car, receive messages, and randomly, after a few seconds, the error.
In arduino, it works ok, no problems.
What else could I try?
I am testing on an Opel Corsa E (2019)
Thanks

ccritix, 2021-03-29:
Looks like I found the solution to the error problem ... at least in my case ...
I added a delay (10) in the Arduino code after line printPacket(&rxPacket);. Looks like there's no problem ... at least testing with a generator. I haven't tested it on the car yet. It follows...

ccritix, 2021-03-31:
Looks like that was the solution for me.
Now I can use the application

closed 2022-01-23 by adamtheone