adamtheone/canDrive #9: Issues with Arduino and MCP2515 CAN Bus Module

fklement, 2021-02-01:
I'm not able to read out CAN Messages with the provided Code and the MCP2515.
I ordered those on amazon: https://www.amazon.de/gp/product/B01MXYYAEN/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I already made the adjustments to match the 8e6 clock frequency.
The code also compiles and get's flashed without any problems.

But I only get:
`CAN RX TX Started`
... on the serial output.

Im currently using a Diamex CAN-Simulator (https://www.diamex.de/dxshop/Diamex-OBD2-Simulator-alle-Protokolle).
There I'm able to select different CAN protocols/speeds. I tested also different speeds and also different MCP2515 modules.

But somehow I'm stuck. Do you have any idea/suggestions for me?

FrankM54, 2021-02-01:
Yesterday I set up an Arduino Nano clone, with the same MCP2515 module and everything worked for me. 

If you are only seeing "CAN RX TX Started" that means the module is initialized but isn't seeing any traffic on the CAN bus.

The Diamex product that you linked to is an "OBD Simulator" and does not broadcast CAN traffic. It needs to be asked for a request and then will provide the data. For example sending 7E0 01 0C will return the engine speed.

What happens when you connect it to a car?

fklement, 2021-02-01:
@FrankM54 Thanks for the response.

So I tested the "OBD Simulator" with a logic analyzer where I only had connected CAN-High and CAN-Low and there I could decode the messages without any problem. That's why I think I should also work just fine when connecting the MCP2515 module.

I will test to see if it behaves the same way when I plug it into my car.

fklement, 2021-02-02:
@FrankM54 Just tested it with my car.

There I'm only receiving the following:
```
recv
10,00,01,2010000000000000
```

I haven't googled it yet. Can you make any sense of it off the top of your head?

FrankM54, 2021-02-02:
Here is a sample from my car (2015 370z)
```
174,00,00,FFEFFEAA05000000
177,00,00,FE100A0000F80000
160,00,00,4C72D60410FBE0
182,00,00,00800000043300B6
174,00,00,FFEFFEAA06000000
177,00,00,FE100A0000F80000
160,00,00,4C72D60410FBE0
182,00,00,00800000043400B6
```

First number is the "arbitration / Module ID" number
Second number is the "RTR / Remote transmission request" request (usually not used in automotive)
Third number is the "IDE /  ID extended"  bit  01 = 29 bit CAN Frame / 00 = 11 bit CAN Frame
Forth numbers are the data bytes / max 8 bytes.

It looks like your vehicle uses a 29 bit / extended CAN frame.
As of right now, the GUI portion does not support 29 bit CAN. frames:https://github.com/adamtheone/canDrive/issues/8#issuecomment-764990022

Also, if you are only seeing one ID, your car might have a OBD gateway.
What is the year, make and model of your car?

fklement, 2021-02-03:
@FrankM54 Thanks for explaining the structure.

Ahh you're right. Totally forgot that fact.
But the arduino part should fully support the extended CAN frame.

Can you explain what an OBD gateway exactly is? I find different things on the internet about "OBD gateways".

My Car:
[image]

FrankM54, 2021-02-03:
From the Arduino output, it is functioning as it should as the data has the proper structure.

I am not very familiar with VW, but it does look like your Mk7 Golf does have a "CAN gateway" 
What it does:
https://www.bosch-mobility-solutions.com/en/products-and-services/passenger-cars-and-light-commercial-vehicles/connectivity-solutions/central-gateway/

VW calls it a "Data bus diagnostic interface -J533"

As I see it, you next step is to locate a wiring diagram so you can understand what modules are on which CAN bus.
Then you will be able to connect directly to the CAN bus you want and see the traffic.

The below pin out of your J533 module might be helpful.
https://www.golfmk7.com/forums/index.php?attachments/canbus-jpg.139389/

Good luck!

fklement, 2021-02-04:
@FrankM54 Thank you so much for the infos!

MohamedAbdelrehem, 2023-10-14:
I have the same problem i think 
I connected my arduino uno with the mcp2515 module to the OBD-II pins 6 and 14 
I don't get any traffic from the car (Chevrolet optra 2010 also called lacetti) 
I am using the example code and changed in the library MCP2515.h the MCP2515_DEFAULT_CLOCK_FREQUENCY to 8e6 based on the crystal oscillator on the MCP2515 Module

adamtheone, 2023-10-15:
You need to narrow down the problem and gather more information. The easiest would be to get an oscilloscope and do measurements: Measure the MCP2515's output to make sure that your Arduino setup works correctly, and measure the OBD's 6 and 14 pins to make sure that your car is communicating on those pins and on others.

Making sure about your arduino setup can also be done with another exact setup (although this is a little overkill and cumbersome), so basically you are connecting: Arduino -> MCP2515 <-> MCP2515 <- Arduino. With this, you could make sure that whatever you send on one end, you can receive on the other. Just make sure the lines are terminated with 120Ohms: There is a jumper on the MCP2515 board that I showed in the video.

Making sure about your car can also be done by searching online and making sure that your car supports the SAEJ1979 standard (High-Speed CAN on pins 6&14), or in turned off status, measure resistance on those pins, and/or in running status, measure the voltage on those pins.

Dodis827, 2024-09-10:
hello I have a golf 7 and I have the same problem I do not receive any data from the control unit, some solved?