adamtheone/canDrive #10: Tried on my 2018 Alfa Romeo Stelvio and it does not work

buglinjo, 2021-02-10:
I have tried this exact code with Arduino Uno and the MCP2515 module on my 2018 Alfa Romeo Stelvio. It says `CAN RX TX Started` in the serial monitor and nothing after that.

The only thing weird that happens when connected is that the start/stop button flashes in my car randomly like it's getting some data :))) 

Speed is 500 I believe as it is an FCA car.

FrankM54, 2021-02-10:
Starting in 2018, FCA added a "Secure Gateway Module" (SGM)
[image]

I don't have the full details, but it is my understanding that the SGW authentication process takes place on the Chrysler servers.

So you can't connect to the CAN bus via the OBD port.... BUT if you use a wiring diagram you can connect to one of the CAN busses AFTER the SGM.

buglinjo, 2021-02-10:
Ohh, that makes sense... Kinda sucks though... Wanted to disable this stupid auto start/stop feature :( That explains why people do not sell OBD port disablers and you need to tap into wires directly. Thank you for the explanation!

buglinjo, 2021-02-10:
This document says that all CAN data is available through the OBD-2 connector. Am I doing something wrong?

http://www.racelogic.co.uk/_downloads/vbox/Vehicles/Other/Docs/Alfa%20Romeo-Stelvio.pdf

FrankM54, 2021-02-11:
I really don't know... 
To verify your setup, try it on a friend's car.

Maybe the VBOX folks know how to authenticate / get through the SGM?

closed 2021-02-11 by buglinjo