adamtheone/canDrive #11: CAN Speed in the activeChannelComboBox CAN_L CAN_M CAN_H How does it work?

rammsteinfanat, 2021-02-23:
Hi! Thank you very much for this work!
Please tell me how this works at Arduino?
I can't understand!
[image]
**in main.py**
```
        if self.activeChannelComboBox.isEnabled():
            txBuf = [0x42, self.activeChannelComboBox.currentIndex()]   # TX FORWARDER
            self.serialWriterThread.write(txBuf)
            txBuf = [0x41, 1 << self.activeChannelComboBox.currentIndex()]  # RX FORWARDER
            self.serialWriterThread.write(txBuf)
```
Sends this
```
for CAN_L [0x42, 0] [0x41, 1] or [66, 0] [65, 1]
for CAN_M [0x42, 1] [0x41, 2] or [66, 1] [65, 2]
for CAN_H [0x42, 2] [0x41, 4] or [66, 2] [65, 4]
```

How does Arduino understand this? And does it work?

I thought it would work here
`CAN.begin(CAN_SPEED)`
Where **CAN_SPEED** can be **(33E3)** or **(95E3)** or **(500E3)**

Sorry for my English

Thank you very mutch!

adamtheone, 2021-02-24:
Thanks Lelick!
That feature in the GUI **does not** work with the arduino code. It will never work because the arduino and the MCP2515 is connected to 1 channel at a time. The speed could be selected through that dropdown, but that's not implemented.
I've used that selector to select between the 3 can busses when I developed my own sniffer hardware that you can see in my first video on youtube. Later, I'm planning to make that hardware available and bring back this feature on the GUI.

Lelick-PMR, 2021-02-25:
Yes. I saw it. 
[image]

I was interested. Thanks Adam for the answer. If I come up with something, I'll write here. If you can change the CAN speed without re-firmware Arduino. You may just have to add a physical switch to change the CAN Speed. 
This
[image] or this [image]
I will try
Sorry about my English. Again.

closed 2022-01-23 by adamtheone