adamtheone/canDrive #12: can use with Realdash soft?

Leechuck2, 2021-03-01:
Hello Adam. Your work is awesome. I have written you comments on youtube but maybe it is better to leave it here.
after watching your videos I bought the arduino, the shield, an ELM327 ... etc and that's when I came across the Realdash program.
With Realdash, totally digital dashboards are created to which you can add all the elements (ABS warning, turn signals, ODO, fuel gauge ... etc) receiving the information via OBD2.
I'm a very newbie and I don't know how to tackle the project. I have seen that you can create a custom XML with the CAN addresses so that the program interprets it and shows you the data. but I don't understand how to convert the addresses that are in the repository with those of Realdash.
I have also seen in the videos that shows the radio information on the mobile, for example, but I have not located those addresses in the repository. it would be nice to be able to add them to the project.
a question. Can messages (text of what Spotify plays, for example) be entered into the CAN system for the CID to display?
As you can see, I have a lot of ideas (which are probably impossible) such as getting the information from the GPS coordinates of the navigation system to use it with Realdash ... well, I find it a very interesting software whose implementation with your work can be beastly!
Thank you very much for your attention.
P.S. I have an Opel Vectra C. I am Spanish and my English is horrible. (I use google XD translator)

adamtheone, 2021-03-06:
First, you should never apologise for your English :)
Let me try to answer your point one by one:
- I've come across Realdash too, and a lot of people seem to be interested in using my tutorials to make Realdash work. Based on this, I'm planning to make some videos of how to do this, but it will take a long time for sure. 
-  You are probably referring to a DBC file. That is for higher-level software, I'm not sure if it would be the right approach to interface my stuff with other software components (e.g. mobile app).
- The mobile piece is still missing, I'm working on it :) 
- I've also tried to display some messages on the CID display, but it's really not trivial, and I still couldn't really figure out how could I do it. But other people have done it, so I'm sure it's possible. (search for Pepas comfort module)
- I've wanted to do these as well, but I'm afraid that the GPS information is not transmitted on the can bus, it's just an internal information of the GPS module (the radio module).

Thanks for reaching out and I'd recommend you to continue learning and build awesome stuff because you clearly have the motivation. I'll try to keep releasing videos with the stuff that we discussed above. Best of luck!

zalexzperez, 2021-03-26:
I've used your amazing python program to hack my Astra H. I'm very grateful for it.

About displaying messages on the CID display, you may want to check out this project:   with the following repo:

zalexzperez, 2021-03-26:
I've used your amazing Python program to hack my Astra H. I'm very grateful for it.

About displaying messages on the CID display, you may want to check out this project based on the Raspberry Pi: https://jb0.de/?p=232 (the project is like 4 entries, not just that page)
Here's the repo: https://github.com/JJToB/MPi3

For what I read, it's only possible to display messages on the A/C field at the main screen. You can write data on other screens, though.

I have a GID display without the stock CD30 radio, so I basically only have the main screen and it's a bummer I can't replace the useless date field taking up 1/4 of the display....which by the way is showing we're in 2007 (again, CD30 not installed in my car)

I wonder if the time can be changed with your Can sniffer program ;D I bought an OPCom device a long time ago but for some reason I was never able to connect with the module/s controlling the infotainment system.

By the way, here's also a repo for decoded CAN Messages on the Astra H: https://github.com/JJToB/Car-CAN-Message-DB

vankaro, 2021-08-23:
Добрый день, начинаю изучать Realdash, пока плохо получается, прошу Вашей помощи, при прослушивании в мониторе данных по CAN шине приходит повторении, например 00 00 05 DE: 00 00 10 00 это норма, но бывает приходит 00 00 05 DE: DE 00 10 00, а в панели Realdash постоянные срабатывания с 0 до максимального значения, как можно избавить от своебразного эхо. Пользуюсь прибором https://aliexpress.ru/item/32968835391.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=40090678953&aff_fcid=f2eb2fada4024b358f02236986756ced-1629721430010-07937-_APOpyu&tt=MG&aff_fsk=_APOpyu&aff_platform=default&sk=_APOpyu&aff_trace_key=f2eb2fada4024b358f02236986756ced-1629721430010-07937-_APOpyu&shareId=40090678953&businessType=ProductDetail&platform=AE&terminal_id=ec918d353048471984f0302a7088b12e&item_id=32968835391&sku_id=66535179530