adamtheone/canDrive #13: CAN-High works but not CAN-Low

Robin-ette, 2021-03-01:
I have an Opel Ampera-e and managed to get data from the CAN-High. I followed the diagrams from your Youtube videos as well as the connection diagram for the "switch-box". Pin 6 to H and Pin 14 to L, on the MCP2515, and 500E3 in the arduino code.

But when I went to read the CAN Low I have tried all different ways. Both connecting the Pins to 1 and 4, as well as 6 and 14. And changing the arduino code CAN_SPEED variable to 33E3. But I get the message , "Starting CAN failed!", both times.

Am I doing something wrong?

Thanks for help.

adamtheone, 2021-03-06:
Hey, 
The low-speed can bus (GMLAN) is not present on all cars and not even all build-years. Have you check that your car has a low-speed can line?

Robin-ette, 2021-03-16:
Hi, thank you for taking the time to answer.
I figured out the problem, I forgot to add the CNF values for the 33E3. 

Works now.

closed 2021-03-16 by Robin-ette