adamtheone/canDrive #16: Receiving only one HEX Code

solai4, 2021-05-28:
i uploaded the code to UNO and i m getting only one hex code.
i tried Engine RPM code and i m getting RPM Values.

197,00,00,0000BE0002000000
197,00,00,0000BE0003000000
197,00,00,0000BE0004000000
197,00,00,0000BE0005000000
197,00,00,0000BE0006000000
197,00,00,0000BE0007000000
197,00,00,0000BE0008000000
197,00,00,0000BE0009000000

adamtheone, 2021-05-29:
This is weird. Can you send some screenshot of the RPM and PID reader examples sketch output?

solai4, 2021-06-01:
> This is weird. Can you send some screenshot of the RPM and PID reader examples sketch output?

Sorry i forget to update. My manufacture is using PIN 8 & 9 for CAN BUS Reading. But i cant read windows up and down but i can able to see the data changes when doors are opened. Im keep on digging .

closed 2022-01-23 by adamtheone