adamtheone/canDrive #19: 4 byte ID corrupts on sending

bionicbone, 2022-01-11:
I do not know if you are still supporting this... I hope so its the best I've seen!!  Great job.

For Info: I have installed the canSniffer on arduino and its running in randomCAN mode. I have installed canDrive on Visual Studio running python 3.9 environment (3.8 not available for VS).

I notice some IDs are extended, like this (OK I dont undertstand why for now but I'm just getting started)

[image]

If we save to the decoded list the extended ID then this is OK

[image]

And we can resend the message and the extended ID is still correct on the bottom part of the screen but the resent message is logged incorrectly.

[image]

[image]

However, when I resend the message I get this, only the last 3 of the ID are logged.

[image]

adamtheone, 2022-01-13:
Hello,
Thanks for the bug report. Its should be a simple fix, I believe it's just just a cropping issue im the GUI. I'll try to see if I'll have some this weekend to check, or you can try to check where that label is updated.

bionicbone, 2022-01-15:
I did some analysis, so far I think it is being cropped in the Arduino either as part of the read or the wirte back.

adamtheone, 2022-01-23:
Hello, 
I think I've found the issue, and pushed the fix into this branch: https://github.com/adamtheone/canDrive/tree/29bit_id_support
The problem was a missing type casting in the arduino code.
Can you test it pls? If it works, I'll merge the fix into the main branch.