adamtheone/canDrive #20: SavvyCAN compatible Session Files

bionicbone, 2022-01-22:
Hi Adam, 
I probably should create a pull request, but I've not got everything set up.
I made a change that you may want to consider, to the save file process that converts to time stamp to ms
Its not pritty, I've not used Python before so its just a logical flow (for me at least)
However, now the file can be loading directly into SavvyCAN for analysis.

def saveTableToFile(self, table, path):
        if path is None:
            path, _ = QFileDialog.getSaveFileName(self, 'Save File', './save', 'CSV(*.csv)')
        if path != '':
            with open(str(path), 'w', newline='') as stream:
                writer = csv.writer(stream)
                for row in range(table.rowCount()-1, -1, -1):
                    rowData = []
                    for column in range(table.columnCount()):
                        item = table.item(row, column)
                        if item is not None:
                            
                            **if column == 0:
                                tempItem = ''
                                for mynext in range(0,len(item.text())):
                                    if item.text()[mynext] !='.':
                                        tempItem += item.text()[mynext:mynext+1]
                            else:
                                tempItem = item.text()
                            rowData.append(str(tempItem))**
                        else:
                            rowData.append('')
                    writer.writerow(rowData)

Once saved load into SavvyCAN using the "GVRET LOGS" format

[image]

[image]

[image]

[image]

[image]

adamtheone, 2022-01-23:
I've pushed the modifications to this branch. Please try it out:
https://github.com/adamtheone/canDrive/tree/session_file_millisec

bionicbone, 2022-01-28:
Leave with me and I will check this weekend.

I suspect further code is required to remove your "human" labels after the ID number for the export.

bionicbone, 2022-01-29:
Can you update the Python like this ?

I have just created many logs for SavvyCan with this code.

    def saveTableToFile(self, table, path):
        if path is None:
            path, _ = QFileDialog.getSaveFileName(self, 'Save File', './save', 'CSV(*.csv)')
        if path != '':
            with open(str(path), 'w', newline='') as stream:
                writer = csv.writer(stream)
                for row in range(table.rowCount()-1, -1, -1):
                    rowData = []
                    for column in range(table.columnCount()):
                        item = table.item(row, column)
                        if item is not None:
                            if column == 0:
                                tempItem = ''
                                for mynext in range(0,len(item.text())):
                                    if item.text()[mynext] !='.':
                                        tempItem += item.text()[mynext:mynext+1]
                            elif column == 1:
                                tempItem = ''
                                for mynext in range(0,len(item.text())):
                                    if item.text()[mynext] !=' ':
                                        tempItem += item.text()[mynext:mynext+1]
                                    else:
                                        break
                            else:
                                tempItem = item.text()
                            rowData.append(str(tempItem))
                        else:
                            rowData.append('')
                    writer.writerow(rowData)

Kwieeciol, 2022-01-29:
@bionicbone I think it'd be much easier if you create a PR referencing this issue

bionicbone, 2022-01-29:
@Kwieeciol it would if I had got GitHub set up correctly

adamtheone, 2022-02-03:
This feature is really specific for SavvyCan, and would mess up other parts of my code: when the program reads back the saved session. This should be either a separate feature from the GUI, or the read back functionality of the GUI should also be aligned with the SavvyCan format. I have very limited availability nowadays, it would be certainly be faster if you can create the modifications and create a PR from it.