adamtheone/canDrive #23: Vectra C - communication every 45 seconds?

chupocro, 2022-02-11:
Szia :-)

I can see we have the same car (I have Vectra C 2.0 2002, motor Y 20 DTH). While I am waiting for the parts to build the device I'd like to ask you if you are by any chance aware of some BUS communication on Vectra C which takes place every 45 seconds?

I am asking because when I start the engine then RPM and speed indicators drop to zero for a moment exactly **every 45 seconds** and when that happens **4 times** then the RPM and speed indicators stay at zero and some warning lights (ABS, alternator, ...) turn on and I can't use signal lights, horn, full beam headlights and the radio turns off.

After turning the engine off and back on everything repeats again and for the next **4 x 45 seconds** everything works well. I removed the CIM and cleaned all the connectors but still the same. When the warning lights are on the error reported by OPCOM is "No communication with CIM".

However, after the engine is on for half an hour or maybe an hour (depends on outside temperature and humidity) then everything start working well.

I removed and cleaned UEC, all fuses and some other connectors and added another battery negative to chassis wire but still couldn't find what is causing the problem. The same problem on many Vectra B 1.6i cars could be solved by just removing the cabin light bulb.

Seems as CIM or ECU is trying to communicate with **something** every 45 seconds and if there is an error then after 4 retries the CIM goes into safe mode. When the CIM is in safe mode hazard lights are still working but there isn't sound of the relay like when everything works well.

In the meantime I tested your GUI program, I still can't test the communication but I can see it's made very well. There is just a small problem, despite the program works well there is a warning:

> QWidget::setLayout: Attempting to set QLayout "" on ModernWindow "", which already has a layout

Are you getting the same warning or the library changed? Seems as t[his could be the solution](https://stackoverflow.com/questions/37304684/qwidgetsetlayout-attempting-to-set-qlayout-on-mainwindow-which-already/37306238).

Since you recomended Python 3.8 I am using Python 3.8.10 in virtualenv + Visual Studio Code.

0hv1, 2022-02-22:
https://google.com

adamtheone, 2022-03-07:
Hello.
Unfortunately, I'm not a car mechanic, so I cannot add anything to the CIM issue.
I have not seeing that issue, but it can be a library version mismatch somewhere. Let me know if you've gottent further with that problem or not.