adamtheone/canDrive #24: Lexus gs450h canbus received but cant send

aljini, 2022-02-27:
Hi there i have used your code and the interface 
I can receive canbus but when i send some back nothing happens 
Close the interface i can see on pycharm teminal ( wrong can package tx started )
I dont know whats the issue here

aljini, 2022-02-27:
[image]

aljini, 2022-02-27:
[image]

aljini, 2022-02-27:
[image]

SubhranshuSharma, 2022-03-02:
[image]
same

SubhranshuSharma, 2022-03-02:
[image]
Same

SubhranshuSharma, 2022-03-02:
[image]
Same

aljini, 2022-03-02:
So its not working with you either did you try to send some messages back ?

SubhranshuSharma, 2022-03-04:
If you mean sending packets into can bus, then still not successful

adamtheone, 2022-03-07:
Hello.
We'll need more information to debug this. The worked for several other people, so I suspect either a hardware problem or something else. First, I highly recommend to try out the CAN examples codes just with the arduino. With that, you can make sure that your connection are fine and bus is working properly. Then, take a look at the ardunio raw serial output and see if that looks correct or not. Only try to debug the GUI as the last step, as the issue is somewhere before that, probably.

aljini, 2022-03-07:
Everything seems fine in examples can recovering and sending and also i get rpm readings from the car so  on the Arduino side everything is fine 
But i suspect that there is a problem can drivers 
Or in the GUi the problem is i dont know how to debug that 
I really need it to work because i am converting my car to electric and this is process is needed in the conversion
So please help me fix that 
I have attached some photos of what errors i am getting 

[image]

[image]

[image]

[image]

[image]

aljini, 2022-03-09:
Today i found out that i can send some codes and the car recives it but the terminal says wrong paket and the sniffing stops it gives the command and the car recives it and it opens the door lock after that the sniffing stops and i cant recive or send anything to the car unless i disconnect it and connect it again 

Somehow i only got the door lock runing but the gui stops working i dont know why

adamtheone, 2022-03-17:
The arduino serial output proves that it is not a GUI issue, because the packets are already corrupted when they exit the arduino. Are you sure that you are using the correct CAN bus speed?

SubhranshuSharma, 2022-03-25:
> The arduino serial output proves that it is not a GUI issue, because the packets are already corrupted when they exit the arduino. Are you sure that you are using the correct CAN bus speed?

yes the can speed is 500e3 i know that as the engine rpm example works fine for me, is it possible that the car(i10 2013 in my case) doesnt have the can bus responsible for core functionality in the obd2 plug, is it possible that the cars brakes/stearing cant be controlled by the can bus on obd2 plug, another possibility is that the 500000 baud rate is not enough for 500e3 can speed so during python and arduino communication  some packets are missing, hence wrong package error.

adamtheone, 2022-03-29:
I have no issues with 500mbaud, so I don't think that it is a bug in my code. It's absolutely possible that you cannot access certain modules in your car, a lot of newer vehicles have "firewalls" on the can bus to protect those devices from attackers. I'm not sure if this is the case in your situation.