adamtheone/canDrive #30: Longan CANBED Dual denies to cooperate with the Arduino code...

YKantBG, 2022-06-01:
Hi Adam et al., 
I have one CANBED DUAL  hanging around and decided to put it in something useful. I had doubts, (confirmed), that it won't work with your code without some intervention and augmentation of the code. Tweaked the MCP2515.h to reflect the CS and INT pins, to no avail, but, the CANReceiver example from the CAN library worked no problem (In fact, this is the only example that actually worked). Actually, the GUI's terminal window also started receiving the packets prefixed with WRONG PACKET. but only when the CANReceiver example is running!?
Would it be too much to ask, you or anyone here, for some pointers where to look at in order to make them mate and work as intended? I am running out of ideas and frustrated with code review and tweaking...

Thanks...

adamtheone, 2022-06-02:
Hello.
I cannot promise anything, but can you shed some more light on the exact setup that you are using now, and the exact goal that you are trying to achieve?

YKantBG, 2022-06-02:
Adam, thanks for the prompt response. So, my setup is super simple. My intention is to use the named board, Lonhan CANBed DUO - it has MCP2515 and MCP2517FD providing dual channel CAN interface., both controlled through a separate CS pins and common INT. Nothing fancy, just have this board from another project and now I am trying to use it to capture all PIDs from two specific vehicles and use those PIDs in a device I am developing to install there. In the past I was capturing everything from the serial interface and it was frustratingly slowly to filter out what i needed. Your application is a gift from heaven to accomplish my task. 
Now, the board works just fine with it's original libraries, both in 2.0 and FD modes. It also works fine in Receive mode with the CAN examples. However, nothing else seems to be working. I believe the link breaks somewhere in the onCANReceive function!?
For your reference, i will attach a link to the board's info page, where all the libraries are also listed (https://docs.longan-labs.cc/1030019/). 
So, long story short - I am trying to utilize a different board to sniff a CAN bus...:(

Cheers!

YKantBG, 2022-06-06:
[spi.txt](https://github.com/adamtheone/canDrive/files/8842253/spi.txt)

adamtheone, 2022-08-14:
Hey, 

So sorry for the late reply, I was super busy with private stuff. Have you managed to solve the problem by now?
I'd have suggested to try to isolate the problem by taking step by step measures. For example, first, we'd want to make sure that the CANbed is indeed sending data through the serial interface. Just open a serial terminal, and see the raw data. This data must be in the format that the GUI is expecting. (The exact format is explained well in my Arduino code I hope.) If that works, try out the GUI in slower baud rates. It's possible that the CANbes is sending too much data at the same time, and GUI is just unable to handle it. Also, if you had an arduino lying around somewhere, I'd definitely try that with the GUI too.

Good luck!

gremixas, 2023-11-20:
CANBed Dual does NOT have MCP2515 on the board. CANBed Dual uses the GD32E103CBT6 as the CAN controller.
Their own made "controller" made my headaches worse. It's nothing but pure garbage when it comes to real canbus connectivity situation. It can handle only low traffic but nowhere near today's car's canbus. Oh, and their library has bugs :D I corrected one that would not let it receive anything shorter that 8 byte long... What a waste of time and money. I now have their CANBed RP2040 with MCP2515 and this one works flawless. But it's only one channel, pitty.