adamtheone/canDrive #31: No data being received with alfa romeo 147

drugward, 2022-08-25:
Hi, I’ve downloaded everything required and set up my arduino as you did in the video.
When arduino is in random data mode i get the normal packets as i should, but when i change it to the mode where its supposed to receive data from the car it displays 0 information.

the car runs on 500kbps and I’ve tried all the speeds, when I select mid speed the can system of my car crashes completely, requiring me to turn it off, update the speed in arduino and then turn it back on.

im just wondering why no data is being received at all, if i use canreceiver from the examples i get nothing again, when using some other examples from other libraries i can get it to send data and crash the canbus once again but no data is received 

even tried another arduino, another board, all for nothing, a friend of mine made a module with arduino nano and mcp which does needle sweep and shows digital speed on the dash and I tried sniffing that data aswell, directly connecting both the modules without the car,connecting it to the car and trying like that also resulted without any data shown in the gui.

WHAT AM I DOING WRONG LOL.

adamtheone, 2022-09-27:
Sorry for the late reply. I'd try to isolate the problem, and wouldn't try to bring the GUI in the picture without making the CAN examples work flawlessly first. It seems that you've tried out a lot of things without success. I assume all the connections are right, and you are sure that your model has CAN on those specific pins, and your car doesn't have CAN firewalls, or such fancy thing. If you are really clueless, I'd take an oscilloscope and just check what are on those funky data lines.
Cheers,

cgavaller2, 2023-01-22:
What Arduino are you using? Also, make sure supply voltage to the MCP2515 board is 5v, not 3.3

boberolak, 2023-01-22:
Hi, sorry, its not my library, i just mirrored it. You have to search for
oryginal library/ author page :/

GiuanniSomething, 2023-04-16:
Hi! I have a smiliar problem with an alfa romeo 156 (1.9 JTD from 2005). I wired the CAN H and CAN L to the correct pins of the mcp2515, and also double checked the SPI connections between my arduino UNO and mcp2515. Whenever i open the serial communication table, it says that the connection started but if i turn on the dashboard or even start the engine i don't receive any kind of data. I've also tried all speeds. Did you manage to find a solution? If so, could you tell me how you did it? Thanks in advance!

closed 2023-04-16 by drugward

GiuanniSomething, 2023-04-16:
Hey, thanks for the really fast reply! As i said before, i've tried all speeds, and on 50kbps i didn't receive any data either. I think that the car uses a standard OBD2 interface because it is the restyling version, the Giugiaro to be exact (i know for sure that the first series didn't use the OBD2 until a year which i don't remember). It uses a BOSCH edc15 ECU, what flips me out is that why wouldn't it use a standard canbus communication? The main informations i need are RPM sensor, Speed sensors and MAP sensor. Can i ask how did you connect to the OBD2 plug? Did you use the MCP2515? Thanks again in advance!

drugward, 2023-04-16:
> Hey, thanks for the really fast reply! As i said before, i've tried all speeds, and on 50kbps i didn't receive any data either. I think that the car uses a standard OBD2 interface because it is the [...]

I've just checked, 156 does not use canbus at all, uses a proprietary serialized connection between modules, some of those being ecu and tcu, which you dont have since its not a selespeed car, your next bet is to try and communicate over the K line using KWP2000/KWPFast protocol.

GiuanniSomething, 2023-04-16:
Thanks again for the answer. I also used two different diagnostic systems, Texa and Launch. The newer one is Texa (the last version of it). Both diagnostics tell me that they are communicating over CAN Bus communication

adamtheone, 2023-04-18:
The timing is interesting, as I had this problem come up again regarding another project of mine. The client is indeed unable to communicate with his Alpha through CAN, but K-Line seems to be working fine (KWPFast).
@GiuanniSomething By any chance, are you able to hook up an oscilloscope to the CAN lines to see what's happening there? Weird that those tools report CAN bus comms.

GiuanniSomething, 2023-04-19:
Hi! @adamtheone, regarding the Alfa 159 i know it works with canbus, but it is strange that your client is having issues with it, considering also that it was produced between 2005 and 2011, you could try with the method i tried the last evening if you didn't find anything on the OBD2 connector. The last evening i tried with a friend of mine to remove the ECU, which is a BOSCH EDC15C7, because i read online that on pin 82 and 83 of the bigger connector i would find the CAN Bus pins. I took a multimeter and measured the resistance between these two pins, i didn't read any resistance. I tried this because i researched a lot and found out that the Alfa romeo 156 with cruise control (just like the one i'm working on) uses CAN communication between the Engine ECU and the ABS ECU, because the ABS ECU must tell the Engine ECU the vehicle speed. The ABS isn't at hand, it's really hard to work on it due to the lack of space, so i went for the engine ECU, and didn't find the can bus pins. On the OBD Connector the pins dedicated for the canbus connection are empty. This drives me crazy, why would such an expensive diagnostic system tell me that there's a communication over can bus? Where are its wires??

adamtheone, 2023-04-19:
Thanks for your comment! It is indeed frustrating.