adamtheone/canDrive #32: No CAN packets incoming in random can mode off

Shavo650, 2023-01-16:
Hi. Thank you for this sniffer. It's only one which i found and it works... almost :) So i made .exe file using your python code, because I can't install python at my laptop (i couldn't fix this so i gave up). App works but only if i upload sketch with #define RANDOM_CAN 1. In this mode i get true packets from car and these generated ones. But I can't send any packet back to the car. In RANDOM_CAN 0 i don't get any packets at all. My Arduino works fine, i tested it with examples and my own code too.

cristianku, 2023-01-16:
same here to me

cristianku, 2023-01-16:
maybe we need a delay also for RANDOM_CAN 0

// Main
void loop() {
  RXcallback();
#if RANDOM_CAN == 1
  CANsimulate();
  delay(100);
#endif
}

cristianku, 2023-01-16:
My communication is working using an other program ( https://github-com.translate.goog/ludwig-v/arduino-psa-diag/tree/master/arduino-psa-diag?_x_tr_sl=fr&_x_tr_tl=it&_x_tr_hl=it&_x_tr_pto=wapp )  for Peugeot.

I have set this can sniffer the same, 500k Can bus speed, clock 8k... but its not showing incoming sniffed packages.

adamtheone, 2023-01-24:
So you guys are saying that the arduino and the wiring works for sure, because you can send and receive packets with the CAN library examples? And, with RANDOM_CAN 1, if you open the serial port within the ArduinoIDE you see the generated messages AND some real packets from the car as well, but with RANDOM_CAN 0, nothing? This is pretty weird.