adamtheone/canDrive #33: No data after successful connection

cgavaller2, 2023-01-22:
I'm seeming to have a similar issue to others:

When connected to my 2012 Ford fusion, all CAN library examples work well, and on start of the CanSniffer.ino code, the "rx tx started" message appears with no errors, but within the GUI application there are no packets visible. No packets through the serial monitor either. The CanReciever example built-in to the library works fine on the same 500Kbps speed, so it isn't that. Any ideas?

adamtheone, 2023-01-24:
Let's concentrate on the arduino code only until you see some packets on the serial monitor. You could try adding some delay in the main loop or somewhere else. Maybe there are too many messages coming in and the arduino just crashes. Let me know.

Stefano19-Bongiovanni, 2023-09-16:
Did you solved it? I have the same problem with my Mercedes Benz a170, when I connect it I receive "CAN RX TX Started
" but no other message. Just minding if I need something extra to do like sending a first message to start receiving, like "hey I'm here, start sending!"

cpocol, 2023-09-18:
Try printing debug messages via serial. Don't use the GUI app but watch what happens in serial monitor. Try to conclude whether the arduino crashes.

Stefano19-Bongiovanni, 2023-09-18:
I tried, the message "CAN RX TX Started" is from the Arduino IDE's serial monitor.

cpocol, 2023-09-18:
Print more :)

adamtheone, 2023-09-19:
As mentioned the obvious above: check your connections, check your oscillator frequency, and check the CAN library examples.