adamtheone/canDrive #35: Still receiving packets after unplugged from the car

DVJocelynShannon, 2023-04-20:
I have RANDOM_CAN = 0 so from what I understand there shouldn't be random data coming through. If I start sniffing while it is plugged into the car then the packets come through in piles,  but when it is unplugged there are still packets coming through.

Is there a reason for this or something I am doing wrong with the setup/code?

adamtheone, 2023-04-20:
Is the data different (rubbish) after unplugging? If yes, it might be just noise, because your terminals (CAN-H and CAN-L) are floating. If it is similar to what you've seen before, it can only be buffered packets still in the Serial pipe, but I haven't experienced this before. Even if they are just buffered packets, they should end pretty soon after unplugging.

cpocol, 2023-09-29:
Make sure your laptop is not in power saving/low performance mode. Currently, as the GUI app is not optimized for speed, a slow laptop can't keep the pace. I can easily observe this although I have a pretty good CPU: i7-10850H

cpocol, 2023-11-01:
You can try my lastest version on https://github.com/cpocol/canDrive
I've made lots of optimizations until I have no more lag, even when I set the CPU performance as low as this:
[image]