adamtheone/canDrive #37: GUI on Silicone Mac

gateseee, 2023-06-06:
Is it possible, to use it somehow on Apple Silicone Mac (MBP 2023 M2 Pro). 
When installing the depencies i am getting errors relating to PyQT5. 

It seems like PyQQT5 is not supported on Silicone Mac's. 

Does someone have a working/valid workaround? 

Br

adamtheone, 2023-06-07:
Personally I don't have experience with Macs, but you should look for answers on PyQt related forums. Or try it with PyQt6, maybe that works.

closed 2023-06-07 by adamtheone

gateseee, 2023-06-08:
Thanks for the fast response. Usually im also not that much into Macs, but i recently got one for developing outside the house thats why i have to struggle with such things now. On my Windows Laptop/Desktop everything works fine.

I will try to get it to work somehow, and will post the solution here, if i am able to get it to work.