adamtheone/canDrive #40: problem with the rpm counter example

superpeg00, 2023-11-02:
hi, i connected the system to my motorbike: a triumph street triple 2022.
i have an external O2 sensor that i log with arduino and print the readings on excel; i would like to plot also the rpm on excel, so i decoded the can message that contain the rpm value (518 00 00 08 D0 D1 00 00 00 00 00 00).
the rpm value is (D1*256+D0)/4 and the id is 518.
I tried to use the rpm counter example of the library changing the code from Can.write(0x0c) to Can.write(0x0518) but it does not work and the code can not exit from the while cycle.
any idea of how can i fix the problem? 
thank you very much.

adamtheone, 2023-11-03:
Hello!

The EngineRPM example works in a different manner. It is not just listening to a certain CAN packet but does a standard **query** (https://en.wikipedia.org/wiki/OBD-II_PIDs#CAN_(11-bit)_bus_format).
This means the following:
Sending a packet to the ECU (0x7DF) with a request of 'sending' the RPM back (0x02 0x01 0x0C), where the response packet will have an id of 0x7E8.

Correct me if I'm wrong, but I think your triumph doesn't comply with this standard, it just sends the RPM value continuously with that ID, no? If that's the case, you don't need to worry about querying, just 'parse' the data and you are good to go. You can just use the CANReceiver example, and maybe put a filter (CAN.filter(0x518)), so you only see the RPM message being received.

Adam

superpeg00, 2023-11-03:
Hello, I used the canreciever example and it works! The only problem was that it did not work with char, so I used float and now it works perfectly.
I had to calculate the rpm so I collected the data bytes of the 0x518 string in a vector(pack[8]) and than I used it to calculate the rpm doing rpm=((pack[0]*256)+pack[1])/4.
Is there a way to access directly the data from the can string?

adamtheone, 2023-11-03:
Your method seems to be correct, this is how it should be done.
I'm happy that it worked!