adamtheone/canDrive #42: HW BOM

SquirrelTactics, 2024-05-14:
Hello I tried to view the BOM but half the stuff is missing and I'm not that good with electronics to be able to figure out what needs to be added and what size. 

I was wondering if I could get some assistance with creating a BOM preferably for for JLPCB so I can have the board assembled by them but if that is not possible then just a complete BOM would be amazing thank you for the hard work that you have put into this.

sheerIok, 2024-12-27:
> Hello I tried to view the BOM but half the stuff is missing and I'm not that good with electronics to be able to figure out what needs to be added and what size.
>
>
>
> I was wondering if I could get [...]

Hi,
Did you find anything about the BOM?

adamtheone, 2025-01-06:
Sorry for the late reply. Unfortunately, I couldn't find the BOM, but it should not be too difficult to find the parts as all of the part numbers are there in the PDF. And, if you can open the project with Altium Designer, the footprints should also help you.