adamtheone/canDrive #43: Metadata Generation Failed

Lizzit, 2024-06-11:
I'm trying to install the dependencies for the GUI and i found out that there is a problem installing the pyqt5-tools library. Each and every time i try to install it, it send me the following error:

```
 error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      pyproject.toml: line 7: using '[tool.sip.metadata]' to specify the project metadata is deprecated and will be removed in SIP v7.0.0, use '[project]' instead 
      Traceback (most recent call last):
        File "C:\Users\Elia\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\Elia\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main   
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Elia\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\sipbuild\api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\sipbuild\abstract_project.py", line 92, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\sipbuild\project.py", line 587, in setup
          self.apply_user_defaults(tool)
        File "C:\Users\Elia\AppData\Local\Temp\pip-install-mv34ku4o\pyqt5_8885985acfb84385bd01e038c77ae524\project.py", line 63, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\pyqtbuild\project.py", line 51, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\sipbuild\project.py", line 237, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "C:\Users\Elia\AppData\Local\Temp\pip-build-env-op7nzl8h\overlay\Lib\site-packages\pyqtbuild\builder.py", line 50, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

###SPECS:
+ Windows - 11
+ Python - 3.12.2
+ PyQt5 - 5.15.10

markkthemk, 2024-06-11:
Using Python 3.11 should fix this issue. Adam used 3.8 during development.

Lizzit, 2024-06-11:
ok, but with the two version of python installed how can i call the pip of the 3.11 instead of the pip of the 3.12?

Lizzit, 2024-06-11:
and also, now that i have installed everithing correctly it throws me another error:

```
DeprecationWarning: sipPyTypeDict() is deprecated, the extension module should use sipPyTypeDictRef() instead
  class canSnifferGUI(QMainWindow, canSniffer_ui.Ui_MainWindow):
<class 'AttributeError'> module 'serial' has no attribute 'tools' <traceback object at 0x000002918DAD68C0>
Traceback (most recent call last):
  File "d:\Programmazione\obd_car\02_canSniffer_GUI\main.py", line 530, in <module>
    main()
  File "d:\Programmazione\obd_car\02_canSniffer_GUI\main.py", line 511, in main
    gui = canSnifferGUI()
          ^^^^^^^^^^^^^^^
  File "d:\Programmazione\obd_car\02_canSniffer_GUI\main.py", line 78, in __init__
    self.scanPorts()
  File "d:\Programmazione\obd_car\02_canSniffer_GUI\main.py", line 493, in scanPorts
    comPorts = serial.tools.list_ports.comports()
               ^^^^^^^^^^^^
AttributeError: module 'serial' has no attribute 'tools'
```

adamtheone, 2024-06-12:
I'd suggest using a virtual environment (venv), or have only that python version installed, or search solutions for pyqt5 in the relevant forums/threads.

closed 2024-06-12 by adamtheone