adamtheone/canDrive #47: Not recieving data from car controlls

nyestle, 2024-08-31:
I am receiving data when sniffing, but I don't get any new data when, for example, I lower the windows or control other functions in the car.

The RPM example worked, so I assume the baud rates are correct, and something is coming through when I start your program.

Could it be possible that my vehicle doesn't control these functions through the CAN bus?

Ford Maverick/Escape 2005.

Or do you have any other suggestions?

Thanks in advance! Great Software by the Way!!

Wimboson, 2024-09-02:
Hello, it is possible that the OBD2 port in your car is connected to a gateway and not directly to the CAN-bus.
I had the same problem with my Mercedes Sprinter W906 2018. 
After connecting directly to one of the CAN buses it worked. 
The speed must then be adjusted to the respective bus.

ejot-eprohaszka, 2024-09-03:
@Wimboson Thanks for the suggestion! I will definetly try that and give feedback.

IMAY2009, 2024-09-06:
I have the same issue on the new VOLVOs. The CAN bus on the OBDII is only for diagnostics. 
Another important point.  There are two CAN BUS networks on new VOLVOs and the comfort CAN BUS (CCM, SEAT & DOOR ECUs and CEM,) all operate at 500kbs.

adamtheone, 2024-09-09:
Others said it right, it can be a gateway issue, it can be a different (maybe separate) CAN bus, or something completely different than CAN bus messages. If you want to be hardcore and figure this out, I'd disassemble the door panel and see if you have a controller in there or not. Then, an oscilloscope could be hooked up to the wires if there's a module.