adamtheone/canDrive #50: Can receive packets but can't send

AriesBreath, 2024-11-07:
Hello. I followed your tutorials on YouTube, I'm working with an Arduino Nano + MCP2515 8MHz. I built your GUI, used Sandeep Mistry library with your modification on the MCP2515.cpp and .h files and open your canSniffer.ino and I'm correctly reading at 500Kbps both in the Arduino IDE serial monitor and inthe GUI, the packets comes with no problem.
[image]

I already mapped some packets in my electric motorcycle Ovaobike MCR-S. The problem comes when I try to send a frame because nothing happens. I tried almost all the decoded packets to no avail. If I try to keep sending them continuously sometimes something changes on my bike screen and then it returns as before, but nothing more. On the terminal I see "wrong packet!" and sometimes it happened something like "no permission to write".

In this example you can see I'm trying to send 3 packets (I even tried with 2 or 1 only).
[image]

Here the message screen ready to push the packets. The ID 606 at D1 changes autonomously sometime so is marked as new packet.
[image]

Here I pushed the packets. Only 217 at D2 changes and not often despite the 100ms delta. 
[image]

While the terminal returns errors
[image]

I have no idea how to solve this, can you help me?

AriesBreath, 2024-11-12:
I should add that I think that the system is not workind due to the constant resending of the legit frames that override my frames. I still haven't found a solution.

adamtheone, 2025-01-06:
Hey there, so sorry for the late reply!!
You are correct with your last assumption: the legit frames are most likely overriding your frames. This happened to me exactly like that when I tried to display some text on the built-in GUI in my car. I haven't found a workaround for this. One (pretty dirty) idea could be to be a middleman between the sender (the radio in my case), and the receiver (the screen) and just don't forward the frames that you want to override. Not sure how reliable or doable this would be.
Regarding the wrong packet message: it is printed, because your arduino reboots probably constantly (as it prints CAN RX TX Started as well). One final tip: if you are unsure about sending, narrow down the problem, and send a hard-coded frame from the arduino directly, not from my GUI. This is to ensure that the issue happens on lower levels.

AriesBreath, 2025-01-07:
Thank you for your reply. For what I learned through the internet there's no way to avoid this problem unless I use a Man In The Middl Attack. That would mean getting a microcontroller before the main vehicle control unit, check every frame and filter what I don't want. Given that the vehicle control unit controls the bike behavior and that whatever malfunction it could have could be disastrous, I gave up trying to write and instead relied on the read only. Thanks anyway for the help and for your guide

closed 2025-01-09 by adamtheone