adamtheone/canDrive #52: Problem finding the MS-CAN bus on my car

DanteCecchetti, 2025-02-04:
Hello,
From my research, all buses except the HS-CAN are not standardized, and the SAE J1962 standard does not specify their location within the OBD2 port. I would like to ask: how can I determine the corresponding pins for these buses in my car?
In my case, I have a Renault, and while searching online, I found an image that specifies the location of the LS-CAN. However, I am a bit confused, isn't the LS-CAN a single-wire bus? Additionally, the image does not specify which pin corresponds to LS-CAN H and which to LS-CAN L.

[image]

adamtheone, 2025-02-07:
Honestly, I could only suggest online research. That's how I found out my pinout as well. Good question about the LS bus. If it's CAN, it's always 2 wires, but as you may have probably seen in my case, the other wire is simply ground. Not sure whether in your case, one of those LS can pins is High, and maybe ground is Low. Or maybe one of them is High and the other one is Low as you mentioned. I'd say the best approach would be to attach an oscilloscope / logic analyzer to those lines and check for yourself. If you don't have them, you can just trial and error with a CAN receiver.

cpocol, 2025-02-07:
@DanteCecchetti What kind of information is available via LS-CAN in your car?

DanteCecchetti, 2025-02-09:
Hello, thanks for your response. One idea I had for finding non-standard buses was using a voltmeter and measuring the impedance with the car turned off (which should give me 120 ohms). From what I’ve researched, the MS bus is generally positioned on pins 11 and 3 of the OBD2 port. Although it is not standardized, many vehicles use it this way. This allows me to confirm that it is a CAN bus; then, I just need to determine which pin is high and which is low, but I can figure that out through trial and error.

On another note, @adamtheone , I wanted to ask if your code is commented, since on GitHub there are no comments explaining what the various functions do. I ask because I am working on my final project for my university degree, which is a reader very similar to yours but with additional functionalities. I wanted to know if your code is commented to help understand the CAN bus logic more easily. I have already read the ISO 11898 standards, but if you have it, it would be very useful.

As for your question, @cpocol , I don’t know exactly what information I will find on my vehicle’s LS CAN bus, probably data related to the car's locking system and that type of information. Regarding my project, we only have the final design of the circuit board so far. The coding and testing still need to be done using an experimental prototype before we proceed with the final circuit, which will go through a review process.

@adamtheone, if you have any additional advice besides what you already provided in the video, I would really appreciate it.

cpocol, 2025-02-10:
The CAN bus has one or more CAN tranceivers. If one of them has the 120 ohms resitor, I measured a 120 ohms. If two of them have the resisors then I measured 60 ohms and the system has worked. I know this is common knowledge, but I wanted to make sure that if you don't measure 120 ohms, it doesn't mean that it's not okay; the only problem is when there is no resistance at all (you measure infinite impedance). Also, I measured the impedance with the car battery disconnected. Before you disconnect the battery, make sure you know the "radio code" :)

georgebv1, 2025-03-31:
[rdcan_2025-03-31_21-25-49.csv](https://github.com/user-attachments/files/19539826/rdcan_2025-03-31_21-25-49.csv)

Hello , I have an opel antara , I use usb can analyzer ... This is my file read with Can Monitor.... I can't get the left signal and the right signal to work

cpocol, 2025-03-31:
Hello, from my (not-so-much) experience, I think, depending on the actual car, some messages are there only to **inform** about some events; sending them back would not act as commands to the car. E.g. if you send the message for the left signal via OBD port will not actuate the left signal. I guess Adam is lucky to have a more hackable car. Somebody else here: please correct me if I'm wrong.